myriadrf/gr-limesdr #1: Closing the flowgraph hangs Gnuradio

gasparka, 2018-04-25:
Closing from the X button.
Following message loops for about 10 sec:
```bash
##################
libusb: warning [add_to_flying_list] failed to arm first timerfd (errno 9)
Write(64 bytes) failed
libusb: warning [add_to_flying_list] failed to arm first timerfd (errno 9)
Write(64 bytes) failed
libusb: warning [add_to_flying_list] failed to arm first timerfd (errno 9)
Write(64 bytes) failed
ERROR: Write(64 bytes) failed

##################
libusb: warning [add_to_flying_list] failed to arm first timerfd (errno 9)
Write(64 bytes) failed
```

yakorev, 2018-04-26:
Hi, gasparka!
I have the same problem, but i noticed that selecting WXGui in the flowgraph doesnot get error when i closing it. The output is like in tutorial "Disconnected from device number 0" 😃

gasparka, 2018-04-26:
I found that 'Kill Graph' works when using QT GUI.

yakorev, 2018-04-28:
'Kill Graph' for me works too, but it does not perform stream close and device reset routines, as said in https://wiki.myriadrf.org/Gr-limesdr_Plugin_for_GNURadio

ztamosevicius, 2018-05-03:
Fixed in commit a283ec413b029eedee3f19952286bdcbbc79c34f

closed 2018-05-03 by ztamosevicius