myriadrf/gr-limesdr #17: AttributeError: 'module' object has no attribute 'source'

1am, 2018-09-08:
Hello,

I've tried testing the gr-limesdr plugin, have all the required utilities but for some reason it throws a Python error when using the [FM_receiver.grc](https://github.com/myriadrf/gr-limesdr/blob/master/examples/FM_receiver.grc) example.

```
Traceback (most recent call last):
  File "/home/../gr-limesdr/examples/top_block.py", line 232, in <module>
    main()
  File "/home/../gr-limesdr/examples/top_block.py", line 226, in main
    tb = top_block_cls()
  File "/home/../gr-limesdr/examples/top_block.py", line 149, in __init__
    self.limesdr_source_0_0 = limesdr.source('device-serial',
AttributeError: 'module' object has no attribute 'source'
```

I'm using GNU Radio Companion 3.7.11 on Ubuntu 18.04 with Python 2.7.15rc1. Do you know what could cause this? It doesn't seem like an issue with hardware or GNU Radio but maybe a version mismatch? Maybe the one I have expects blocks to follow a different API.

otamulev, 2018-09-09:
Hi, 1am

Did you run "sudo ldconfig" in terminal window after "sudo make install"?

1am, 2018-09-09:
Hi @otamulev 

Yes, I followed all steps of  the installation instructions. The block is found correctly in GR.

otamulev, 2018-09-09:
Please try ty perform uninstall by typing:

sudo make uninstall
sudo ldconfig

Then delete gr-limesdr folder, download and install it again. Do you get the same output? I have seen it happen however it is usually fixed by performing clean install.

1am, 2018-09-09:
Thanks for the suggestion but it ends up with the same error consitently unfortunately

otamulev, 2018-09-09:
Not really. This error basically tells you that GNU Radio is unable to find blocks. What you see inside flow graph is only .xml representation of blocks. Could you share your cmake output during clean installation?

1am, 2018-09-09:
```
➜  build git:(master) ✗ cmake ..
-- The CXX compiler identification is GNU 7.3.0
-- The C compiler identification is GNU 7.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Configuring Boost C++ Libraries...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   system
--   chrono
--   date_time
--   atomic
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Configuring GNU Radio C++ Libraries...
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so  
GNURADIO_RUNTIME_FOUND = TRUE
-- Configuring LimeSuite C++ Libraries...
-- Checking for module 'LimeSuite'
--   Found LimeSuite, version 18.04.1-myriadrf1~bionic
-- Found LimeSuite: /usr/include/lime, /usr/lib/x86_64-linux-gnu/libLimeSuite.so
-- 
-- Checking for module SWIG
-- Disabling SWIG because version check failed.
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.15rc1", minimum required is "2") 
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.15", minimum required is "2") 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/.../gr-limesdr/build

➜  build git:(master) ✗ make
Scanning dependencies of target gnuradio-limesdr
[ 16%] Building CXX object lib/CMakeFiles/gnuradio-limesdr.dir/source_impl.cc.o
[ 33%] Building CXX object lib/CMakeFiles/gnuradio-limesdr.dir/sink_impl.cc.o
[ 50%] Building CXX object lib/CMakeFiles/gnuradio-limesdr.dir/common/device_handler.cc.o
[ 66%] Linking CXX shared library libgnuradio-limesdr.so
[ 66%] Built target gnuradio-limesdr
Scanning dependencies of target pygen_python_399de
[ 83%] Generating __init__.pyc
[100%] Generating __init__.pyo
[100%] Built target pygen_python_399de
Scanning dependencies of target pygen_apps_9a6dd
[... 22 lines cut ...]

```
➜  build git:(master) ✗ python
Python 2.7.15rc1 (default, Apr 15 2018, 21:51:34) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import limesdr
>>> limesdr.__dict__
{'__builtins__': {'bytearray': <type 'bytearray'>, 'IndexError': <type 'exceptions.IndexError'>, 'all': <built-in function all>, 'help': Type help() for interactive help, or help(object) for help about object., 'vars': <built-in function vars>, 'SyntaxError': <type 'exceptions.SyntaxError'>, 'unicode': <type 'unicode'>, 'UnicodeDecodeError': <type 'exceptions.UnicodeDecodeError'>, 'memoryview': <type 'memoryview'>, 'isinstance': <built-in function isinstance>, 'copyright': Copyright (c) 2001-2018 Python Software Foundation.
All Rights Reserved.

Copyright (c) 2000 BeOpen.com.
All Rights Reserved.

Copyright (c) 1995-2001 Corporation for National Research Initiatives.
All Rights Reserved.

Copyright (c) 1991-1995 Stichting Mathematisch Centrum, Amsterdam.
All Rights Reserved., 'NameError': <type 'exceptions.NameError'>, 'BytesWarning': <type 'exceptions.BytesWarning'>, 'dict': <type 'dict'>, 'input': <built-in function input>, 'oct': <built-in function oct>, 'bin': <built-in function bin>, 'SystemExit': <type 'exceptions.SystemExit'>, 'StandardError': <type 'exceptions.StandardError'>, 'format': <built-in function format>, 'repr': <built-in function repr>, 'sorted': <built-in function sorted>, 'False': False, 'RuntimeWarning': <type 'exceptions.RuntimeWarning'>, 'list': <type 'list'>, 'iter': <built-in function iter>, 'reload': <built-in function reload>, 'Warning': <type 'exceptions.Warning'>, '__package__': None, 'round': <built-in function round>, 'dir': <built-in function dir>, 'cmp': <built-in function cmp>, 'set': <type 'set'>, 'bytes': <type 'str'>, 'reduce': <built-in function reduce>, 'intern': <built-in function intern>, 'issubclass': <built-in function issubclass>, 'Ellipsis': Ellipsis, 'EOFError': <type 'exceptions.EOFError'>, 'locals': <built-in function locals>, 'BufferError': <type 'exceptions.BufferError'>, 'slice': <type 'slice'>, 'FloatingPointError': <type 'exceptions.FloatingPointError'>, 'sum': <built-in function sum>, 'getattr': <built-in function getattr>, 'abs': <built-in function abs>, 'exit': Use exit() or Ctrl-D (i.e. EOF) to exit, 'print': <built-in fun [... cut ...] pe 'exceptions.OverflowError'>}, '__file__': '/usr/local/lib/python2.7/dist-packages/limesdr/__init__.pyc', '__package__': 'limesdr', '__path__': ['/usr/local/lib/python2.7/dist-packages/limesdr'], '__name__': 'limesdr', '__doc__': '\nThis is the GNU Radio LIMESDR module. Place your Python package\ndescription here (python/__init__.py).\n'}
>>> l = limesdr.source
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'source'
```

1am, 2018-09-09:
Ok, looked at the sources of **/usr/local/lib/python2.7/dist-packages/limesdr/__init__.py** and CMake output and there was an issue with my SWIG which seems to be the root of all this. Will check it after reinstalling SWIG

1am, 2018-09-09:
Yes, that was it. Thank you for your time and help @otamulev 

Maybe it would be good to break the build if SWIG fails by throwing a `message( FATAL_ERROR "SWIG not found)` in https://github.com/myriadrf/gr-limesdr/blob/master/swig/CMakeLists.txt#L34 so it's more obvious for people starting in the field?

closed 2018-09-09 by 1am

mcdiarmid, 2018-10-26:
Hi there,

I've encountered the same error that was raised originally in this post.  However, I don't seem to have any issues with SWIG as indicated by the output of `cmake ..` that I have included at the end.  I followed the clean install instructions multiple times with the same issue popping up.  I'm not overly familiar with cmake,  I have only ever used it to install modules for GNU Radio and this is the first time I have encountered any issues.  Any advice would be greatly appreciated!

```
-- The CXX compiler identification is GNU 8.2.1
-- The C compiler identification is GNU 8.2.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Configuring Boost C++ Libraries...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.68.0
-- Found the following Boost libraries:
--   thread
--   system
--   chrono
--   date_time
--   atomic
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.14") found components:  doxygen missing components:  dot
-- Configuring GNU Radio C++ Libraries...
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.5.3") 
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.13.4
 * INCLUDES=/usr/include
 * LIBS=/lib/libgnuradio-runtime.so;/lib/libgnuradio-pmt.so
-- Found GNURADIO_RUNTIME: /lib/libgnuradio-runtime.so;/lib/libgnuradio-pmt.so  
GNURADIO_RUNTIME_FOUND = TRUE
-- Configuring LimeSuite C++ Libraries...
-- Checking for module 'LimeSuite'
--   Found LimeSuite, version 18.10.0-18.10.0
-- Found LimeSuite: /usr/include/lime, /usr/lib/libLimeSuite.so
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig  
-- Found PythonLibs: /usr/lib/libpython2.7.so (found suitable version "2.7.15", minimum required is "2") 
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.15", minimum required is "2") 
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/.../Documents/GNURadio/gr-limesdr/build
```

mcdiarmid, 2018-10-28:
Fixed the issue.  I had to manually add a `.conf` file containing `/usr/local/lib` to `/etc/ld.so.conf.d/` in order for the limesdr module (or any other Out-of-Tree module) to use C++ blocks via SWIG.  I did not have this issue while using Ubuntu, but seem to have encountered it while using Arch-linux based Manjaro.

GuilhermeLopes95, 2020-02-26:
> Yes, that was it. Thank you for your time and help @otamulev
>
> Maybe it would be good to break the build if SWIG fails by throwing a `message( FATAL_ERROR "SWIG not found")` in [...]

What did you do to your swig? I have the same issue