myriadrf/gr-limesdr #22: SetFrequencySXT(1575.42 MHz) - cannot deliver frequency

SpenserCai, 2018-11-27:
In LimeSuite set SXT 1575.42MHz is ok but in gnuradio(gr-limesdr and soapyuhd) will have a 
"SetFrequencySXT(1575.42 MHz) - cannot deliver frequency" error

LimeSuite Sink (TX) info

##################
Connecting to device
##################
Devices found: 1
##################
Device list:
Nr.:|0|device:|LimeSDR Mini, media=USB 3.0, module=FT601, addr=24607:1027, serial=1D423421DE7C61
##################
Device number 0 from the list is used.
Reference clock 40.00 MHz
##################

INFO: device_handler::set_chip_mode(): SISO mode set for device number 0.
SetFrequencySXT(1575.42 MHz) - cannot deliver frequency
INFO: device_handler::set_rf_freq(): RF frequency set [TX]: 1575.42 MHz.
INFO: device_handler::set_samp_rate(): set sampling rate: 10 MS/s.
INFO: device_handler::set_analog_filter(): analog filter channel 0 [TX]: disabled (maximum value set).
Filter calibrated. Filter order-2nd, set to 130 MHz
TX LPF configured
GFIR LPF cannot be set to the requested bandwidth
INFO: device_handler::set_digital_filter(): digital filter channel 0 set [TX]: disabled.
INFO: device_handler::set_antenna(): channel 0 antenna set [TX]: BAND2.
INFO: device_handler::set_gain(): set gain [TX]: 30 dB.
INFO: device_handler::calibrate(): calibration channel 0 [TX]: disabled.
INFO: sink_impl::init_stream(): sink channel 0 (device nr. 0) stream setup done.
---------------------------------------------------------------
Warning: the blks2.tcp_source block is deprecated.
libusb: warning [libusb_exit] application left some devices open

gracid, 2019-02-11:
Does anyone else have this problem? This seems more related to hardware.

giech, 2019-03-23:
I am also having this problem with various frequencies on a LimeSDR-mini on Ubuntu 16.04. Happy to provide more details/logs.

gracid, 2019-03-28:
For a start HW, GW versions and log from gnuradio would be useful. Also make sure you are using newest version of LimeSuite.

giech, 2019-03-28:
Info from `LimeSuiteGui` (v19.01.0, build date 2019-02-08, running on Ubuntu 16.04): `LimeSDR-Mini FW:6 HW:2 Protocol:1 GW:1.29 Ref Clk: 40.00 MHz`

Log from `gnuradio`:
```
LimeSuite Sink (TX) info

##################
Connecting to device
##################
Devices found: 1
##################
Device list:
Nr.:|0|device:|LimeSDR Mini, media=USB 2.0, module=FT601, addr=24607:1027, serial=XXXXXXXXXXXXXX (removed)
##################
INFO: device_handler::open_device(): no serial number. Using first device in the list. Use "LimeUtil --find" in terminal to find prefered device serial.
##################
Device number 0 from the list is used.
Reference clock 40.00 MHz
##################

INFO: device_handler::enable_channels(): SISO CH0 set for device number 0.
INFO: device_handler::set_samp_rate(): set sampling rate: 2 MS/s.
INFO: device_handler::set_rf_freq(): Selected TX path: Band 2
RF frequency set [TX]: 800 MHz.
INFO: device_handler::set_analog_filter(): Filter calibrated. Filter order-2nd, set to 50 MHz
TX LPF configured
INFO: device_handler::set_gain(): set gain [TX] CH0: 60 dB.
INFO: device_handler::set_antenna(): Selected TX path: Band 2
CH0 antenna set [TX]: BAND2.
INFO: device_handler::calibrate(): Tx calibration finished
INFO: sink_impl::init_stream(): sink channel 0 (device nr. 0) stream setup done.
INFO: device_handler::set_rf_freq(): SetFrequencySXT(1500 MHz) - cannot deliver frequency
```

gracid, 2019-03-29:
Does the issue happen with source(RX) block?
Ok, can you disable filters, calibration and lower the gain. If the issue persists can you compile debug version of LimeSuite? You can follow this guide: [link](https://wiki.myriadrf.org/Lime_Suite#Building_from_source) but instead of following step 3.2.1 do this:

```
git clone https://github.com/myriadrf/LimeSuite.git
cd LimeSuite
mkdir builddir && cd builddir
cmake -DCMAKE_BUILD_TYPE=Debug ../
make -j4
sudo make install
sudo ldconfig

cd LimeSuite/udev-rules
sudo ./install.sh
```
(To compile normal version do `cmake -DCMAKE_BUILD_TYPE=Release ../` instead of `=Debug`)
Then post `gnuradio` log again when the issue happens.

giech, 2019-03-29:
I have tried all your suggestions, and the issue persists, no matter what gain value I use, and whether the calibration and filter bandwidth are 0 or 50MHz. The issue also persists whether the LimeSDR-mini is connected over USB2.0 or USB3.0. I am attaching the [debugging log](https://github.com/myriadrf/gr-limesdr/files/3023949/lime.log) as requested.

gracid, 2019-04-03:
Does the same issue happen for RX? Can you give me the flowgraph you are using? Can you do `LimeQuickTest` and post the results here. Can you try setting the specific frequency in LimeSuite, does that work?

giech, 2019-04-03:
I haven't tested extensively with RX, but I have not noticed a problem so far. I think that I have narrowed down the problem a bit. Specifically, the [GnuRadio block](https://github.com/myriadrf/gr-limesdr/files/3038901/tx.txt) seemed functional when I tried it initially, but then: [image]

Reattaching and resetting, I found that some settings are occasionally off for some reason, although I am not sure what causes the issue: [image]

PS
[`LimeQuickTest` before failure](https://github.com/myriadrf/gr-limesdr/files/3038899/test_before.log)
[`LimeQuickTest` after failure](https://github.com/myriadrf/gr-limesdr/files/3038898/test_after.log)
I forgot to take the temperature in the screenshot, but it was 52 degrees Celsius.

gracid, 2019-04-05:
Can you save the ini file from limesuite (before getting that error in first screenshot)? Also, the settings look off because you used `Reset` button instead of `Default`.

giech, 2019-04-05:
In this set of experiments, the failed frequency was 2.8GHz. I first used the `tx.grc` program until `SetFrequencySTX` failed. Then, I went to `LimeSuiteGui` and clicked `Chip->Gui` and `Save`. Is this what you meant? If so, I am attaching [settings.ini](https://github.com/myriadrf/gr-limesdr/files/3049719/settings.txt). I am also attaching the `DEBUG`-level [log file](https://github.com/myriadrf/gr-limesdr/files/3049722/gui.log) when trying to set the frequency on the GUI. You will notice that the command alternates between failing and succeeding, seemingly toggling between `VCOM` and `VCOH` (I was only clicking `calculate`, without making any other changes).

gracid, 2019-06-04:
If it is still possible can you update LimeSuite to the newest version (19.04), connect your board, open "modules->API Calls". Then press  "Run" on LMS_Init, LMS_EnableChannel (set it to TX) and LMS_SetLOFrequency (set it to TX and 1.5 GHz or 2.8GHz). Does the error happen then?

giech, 2019-06-05:
I was unable to reproduce the issue on `LimeSuiteGUI` 19.04 using your instructions. I varied the frequency from 100MHz to 3GHz in steps of 100MHz, but did not get an error.

dgramop, 2019-07-06:
I'm having a similar issue, but LimeQuickTest passes and LimeSuiteGUI works for tuning and FFT. Any ideas?

dgramop, 2019-07-07:
Logs, just in case
Note: Using device: LimeSDR-Mini(1D4C4445A57EF3) GW: 1.30 FW: 6
```
enerating: '/home/dgramop/top_block.py'

Executing: /usr/bin/python -u /home/dgramop/top_block.py

---------------------------------------------------------------
LimeSuite Source (RX) info

##################
Connecting to device
##################
LimeSuite version: 19.04.0-gabf82140
gr-limesdr version: 2.1.4
##################
Device list:
Nr.:0 device:LimeSDR Mini, media=USB 2.0, module=FT601, addr=24607:1027, serial=1D4C4445A57EF3
##################
INFO: device_handler::open_device(): no serial number. Using first device in the list.
Use "LimeUtil --find" in terminal to find prefered device serial.
Reference clock 40.00 MHz
Using device: LimeSDR-Mini(1D4C4445A57EF3) GW: 1.30 FW: 6
##################

INFO: device_handler::enable_channels(): SISO CH0 set for device number 0.
INFO: device_handler::set_samp_rate(): SetFrequencyCGEN(0.128 MHz) - cannot deliver requested frequency

>>> Done
```

gracid, 2020-03-09:
Closing stale issue and most likely a board specific problem.

closed 2020-03-09 by gracid