myriadrf/gr-limesdr #27: gnuradio-companion python blocks are broken (KeyError)

phansel, 2019-01-11:
It is no longer possible to use gr-limesdr blocks in GRC or python.

paul@comp22:~/gr-limesdr/examples$ gnuradio-companion signal_analyzer.grc 
DEBUG:grc:Loading platform
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
DEBUG:grc:Loading application
DEBUG:grc:Running
<<< Welcome to GNU Radio Companion 3.8tech-preview-170-g7333f851 >>>

Block paths:
	/home/paul/.grc_gnuradio
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/paul/gr-limesdr/examples/signal_analyzer.grc"
>>> Converting from XML
Block id "limesdr_source" not found
Error: 'id'
>>> Failure
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/gnuradio/grc/gui/MainWindow.py", line 261, in new_page
    file_path=file_path,
  File "/usr/local/lib/python3.5/dist-packages/gnuradio/grc/gui/Notebook.py", line 107, in __init__
    flow_graph.import_data(initial_state)
  File "/usr/local/lib/python3.5/dist-packages/gnuradio/grc/core/FlowGraph.py", line 381, in import_data
    block.import_data(**block_data)
  File "/usr/local/lib/python3.5/dist-packages/gnuradio/grc/core/blocks/block.py", line 344, in import_data
    self.params['id'].value = name
KeyError: 'id'

Related to this issue: https://github.com/gnuradio/gnuradio/issues/2039

System is Ubuntu 16.04 LTS, GR is compiled with Python3.5.

phansel, 2019-01-13:
To clarify, this is not an issue in GR as of 3.7.9; it is only broken in 3.8tech-preview.

gracid, 2019-06-04:
Main issue for GNU-Radio 3.8 is #44.

closed 2019-06-04 by gracid