myriadrf/gr-limesdr #30: Boot Library Confusion in Cmake

MountainLogic, 2019-01-25:
I'm trying to build gr-limesdr on OSX after upgrading to Mojave the removing all homebrew and port then reinstalling GNURadio from scratch with MacPorts to clean-up tons of conflicts I'd built-up.  I've done the usual git, mkdir build, cmake .. two-step but was receiving an error from cmake:
`
Scott-OSX:build scott$ cmake ..
-- Build type not specified: defaulting to release.
-- Configuring Boost C++ Libraries...
CMake Error at /opt/local/share/cmake-3.13/Modules/FindBoost.cmake:1357 (file):
  file STRINGS file "/usr/local/include/boost/version.hpp" cannot be read.
Call Stack (most recent call first):
  CMakeLists.txt:74 (find_package)

-- Could NOT find Boost
CMake Error at CMakeLists.txt:77 (message):
  Boost required to compile limesdr

-- Configuring incomplete, errors occurred!
See also "/Users/scott/WIP/gr-limesdr/build/CMakeFiles/CMakeOutput.log".
See also "/Users/scott/WIP/gr-limesdr/build/CMakeFiles/CMakeError.log".
`

After reviewing the port boost version:

`
Scott-OSX:build scott$ port info boost
boost @1.66.0_3 (devel)
Sub-ports:            boost-numpy
Variants:             clang33, clang34, clang37, clang38, clang39, clang40, clang50, clang60, clang70, debug, mpich,
                      mpich_devel, [+]no_single, [+]no_static, openmpi, openmpi_devel, python26, [+]python27, python33,
                      python34, python35, python36, python37, regex_match_extra, universal

Description:          Boost provides free portable peer-reviewed C++ libraries. The emphasis is on portable libraries
                      which work well with the C++ Standard Library.
Homepage:             http://www.boost.org

Library Dependencies: zlib, expat, bzip2, libiconv, icu, python27
Platforms:            darwin
License:              Boost-1
Maintainers:          Email: ryandesign@macports.org, GitHub: ryandesign
                      Email: michaelld@macports.org, GitHub: michaelld
                      Policy: openmaintainer

'
After seeing that the boost version is "@1.66.0_3 (devel)", I edited the cmake file at line 64 to read (note the addition of "1.66.0_3" that matches what my version of boot reported):

`
set(Boost_ADDITIONAL_VERSIONS
    "1.35.0" "1.35" "1.36.0" "1.36" "1.37.0" "1.37" "1.38.0" "1.38" "1.39.0" "1.39"
    "1.40.0" "1.40" "1.41.0" "1.41" "1.42.0" "1.42" "1.43.0" "1.43" "1.44.0" "1.44"
    "1.45.0" "1.45" "1.46.0" "1.46" "1.47.0" "1.47" "1.48.0" "1.48" "1.49.0" "1.49"
    "1.50.0" "1.50" "1.51.0" "1.51" "1.52.0" "1.52" "1.53.0" "1.53" "1.54.0" "1.54"
    "1.55.0" "1.55" "1.56.0" "1.56" "1.57.0" "1.57" "1.58.0" "1.58" "1.59.0" "1.59"
    "1.60.0" "1.60" "1.61.0" "1.61" "1.62.0" "1.62" "1.63.0" "1.63" "1.64.0" "1.64"
    "1.65.0" "1.65" "1.66.0" "1.66" "1.67.0" "1.67" "1.68.0" "1.68" "1.69.0" "1.69"
    "1.66.0_3"
)
`
And it seemed to compile with modest warnings:

'
Scott-OSX:build scott$ cmake ..
-- Build type not specified: defaulting to release.
-- Configuring Boost C++ Libraries...
CMake Error at /opt/local/share/cmake-3.13/Modules/FindBoost.cmake:1357 (file):
  file STRINGS file "/usr/local/include/boost/version.hpp" cannot be read.
Call Stack (most recent call first):
  CMakeLists.txt:74 (find_package)

-- Could NOT find Boost
CMake Error at CMakeLists.txt:77 (message):
  Boost required to compile limesdr

-- Configuring incomplete, errors occurred!
See also "/Users/scott/WIP/gr-limesdr/build/CMakeFiles/CMakeOutput.log".
See also "/Users/scott/WIP/gr-limesdr/build/CMakeFiles/CMakeError.log".
Scott-OSX:build scott$ cmake ..
-- Build type not specified: defaulting to release.
-- Configuring Boost C++ Libraries...
-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   thread
--   system
--   chrono
--   date_time
--   atomic
-- Found Doxygen: /opt/local/bin/doxygen (found version "1.8.14") found components:  doxygen dot 
-- Configuring GNU Radio C++ Libraries...
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib;/opt/local/lib/liblog4cpp.dylib
GNURADIO_RUNTIME_FOUND = TRUE
-- Configuring LimeSuite C++ Libraries...
-- Found LimeSuite: /usr/local/include/lime, /usr/local/lib/libLimeSuite.dylib
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found PythonLibs: /usr/lib/libpython2.7.dylib (Required is at least version "2") 
-- Configuring done
CMake Warning (dev):
  Policy CMP0068 is not set: RPATH settings on macOS do not affect
  install_name.  Run "cmake --help-policy CMP0068" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

  For compatibility with older versions of CMake, the install_name fields for
  the following targets are still affected by RPATH settings:

   gnuradio-limesdr

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/scott/WIP/gr-limesdr/build
Scott-OSX:build scott$ 

`

Not sure best practices for cmake, etc.  Am I supposed to force the latest version in port, tell cmake to just live with my version, etc?  Will I regret building from a dev boost, what's the best way to handle this type of issue, etc? Mostly putting this here as a note in a bottle for the next person.  I did see similar issues on my Linux box.  Regardless of my likely self-inflicted install wounds, I'm really enjoying LimeSDR and the new driver, thanks team.  (PS would greatly value control of GPIOs)
-scott

gracid, 2019-02-11:
Well for a start, you should try it with a normal boost version.

ra1nb0w, 2019-03-05:
I pushed `gr-limesdr` on macports. try it ;-)

MountainLogic, 2019-03-07:
Thank you @ra1nb0w ra!nb0w.  The MacPort installed easily on OSX.  Having other issues, but will close this and open another more closely related.

MountainLogic, 2019-03-07:
Able to install from new OSX MacPorts package.

closed 2019-03-07 by MountainLogic