myriadrf/gr-limesdr #34: gr-limesdr on OSX - any hints?

mhaberler, 2019-03-09:
this compiles on macOS Mojave 10.14.13 and gnuradio installed via Macports, but does not show up under grc

the ``make install``part seems to need adaptation as files go to /usr/local

any suggestions?

thanks in advance

Michael

gracid, 2019-03-13:
I'm not familiar with macOS, but to install to /usr/local you could do:
`cmake --DCMAKE_INSTALL_PREFIX=/usr/local ../`
`make`
`sudo make install`
And I don't know if it is needed on mac
`sudo ldconfig`

gracid, 2019-09-05:
I assume the issue is resolved? If it isn't, reopen the issue.

closed 2019-09-05 by gracid

vileer, 2019-10-17:
gr-limesdr install the block define file in the folder /usr/local/share/gnuradio/grc/blocks/ , but the default block paths of gnuradio-companion on OSX is /opt/local/share/gnuradio/grc/blocks, so you have to link all the block define file to this folder. The below command works for me(Mojave 10.14.6).
`sudo ln -s /usr/local/share/gnuradio/grc/blocks/limesdr_sink.xml /opt/local/share/gnuradio/grc/blocks/
`
    
`
sudo ln -s /usr/local/share/gnuradio/grc/blocks/limesdr_sink.xml /opt/local/share/gnuradio/grc/blocks/
`

LRGH, 2022-02-05:
Another approach to use LimeSuite and gr-limesdr on MacOS 10.14.6 which resulted in a working installation for me.

I am using MacPorts to install gnuradio, with wxgui. I chose to stay with GnuRadio 3.7 for stability reasons.

I chose to install both LimeSuite and gr-limesdr without sudo, in $SOURCEDIR, which for example enables me to have multiple simultaneous installations, in case a software upgrade breaks something.

```sudo port install gnuradio37 +docs+grc+log4cpp+logging+python27+qtgui+sdl+uhd+wavelet+zeromq+wxgui

SOURCEDIR=/Whatever/You/Want

cd $SOURCEDIR
git clone https://github.com/myriadrf/LimeSuite.git
cd LimeSuite
git checkout stable
mkdir builddir && cd builddir
cmake ../ -DCMAKE_INSTALL_PREFIX=$SOURCEDIR/LimeSuite/INSTALLED -DwxWidgets_CONFIG_EXECUTABLE=/opt/local/Library/Frameworks/wxWidgets.framework/Versions/wxWidgets/3.0/bin/wx-config
make
make install

export DYLD_LIBRARY_PATH=$SOURCEDIR/LimeSuite/INSTALLED/lib
# Run the following command to validate your LimeSuite installation and that you can connect to the LimeSDR.
$SOURCEDIR/LimeSuite/INSTALLED/bin/LimeQuickTest

cd $SOURCEDIR
git clone https://github.com/myriadrf/gr-limesdr
cd gr-limesdr
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=$SOURCEDIR/gr-limesdr/INSTALLED -DLIMESUITE_INCLUDE_DIRS=$SOURCEDIR/LimeSuite/INSTALLED/include/lime -DLIMESUITE_LIB=$SOURCEDIR/LimeSuite/INSTALLED/lib
make # stops with a linking bug
cd lib; sh lib/CMakeFiles/gnuradio-limesdr.dir/link.txt; cd ..
make
make install

export PYTHONPATH=$SOURCEDIR/gr-limesdr/INSTALLED/lib/python2.7/site-packages:$SOURCEDIR/gr-limesdr/INSTALLED/lib/python2.7/site-packages/limesdr

# add
# local_blocks_path = $SOURCEDIR/gr-limesdr/INSTALLED/share/gnuradio/grc/blocks
# in the [grc] section of ~/.gnuradio/config.conf

gnuradio-companion $SOURCEDIR/gr-limesdr/examples/FM_receiver.grc
```