myriadrf/gr-limesdr #37: gnuradio 3.8 (git) not build gr-limesdr

Red-Owl, 2019-03-11:
cmake (ok)
make ...:
[ 56%] Swig source
/home/gufo/Scrivania/gr-limesdr/swig/limesdr_swig.i:5: Error: Unable to find 'gnuradio.i'
make[2]: *** [swig/CMakeFiles/limesdr_swig_swig_compilation.dir/build.make:64: swig/CMakeFiles/limesdr_swig.dir/limesdr_swigPYTHON.stamp] Error 1
make[2]: *** Eliminazione del file "swig/CMakeFiles/limesdr_swig.dir/limesdr_swigPYTHON.stamp"
make[1]: *** [CMakeFiles/Makefile2:309: swig/CMakeFiles/limesdr_swig_swig_compilation.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
gufo@meo:~/Scrivania/gr-limesdr/bu$

gracid, 2019-03-13:
Currently gnuradio 3.8 is not supported. The supported version is 3.7.13.4 from gnuradio-maint branch.

gracid, 2019-06-04:
Main issue for GNU-Radio 3.8 is #44.

closed 2019-06-04 by gracid

hj1980, 2019-08-23:
FYI, I temporarily worked around this by using the following argument to cmake:
```
-DGNURADIO_ALL_INCLUDE_DIRS=/usr/include/gnuradio/swig
```