myriadrf/gr-limesdr #39: Issue with latest GRC blocks

eokeeffe, 2019-03-12:
Getting this error for some reason, I only just updated the repo and redid the cmake.

Traceback (most recent call last):
  File "top_block.py", line 358, in <module>
    main()
  File "top_block.py", line 346, in main
    tb = top_block_cls()
  File "top_block.py", line 163, in __init__
    self.limesdr_sink_0 = limesdr.sink('', 0, '', '')
TypeError: make() takes exactly 30 arguments (4 given)

Any clues as too what might be happening?

Ubuntu 16.04, GRC 3.7.10.1-62

gracid, 2019-03-13:
It looks like a problem with version mismatch, as it python expects v1.0 xml blocks and you are using v2.0 xml blocks. After redoing cmake did you make and make install?

eokeeffe, 2019-03-13:
I first did a make uninstall, cmake .. , make and then make install

I checked the xml blocks and they match the ones in the updated repo

gracid, 2019-03-13:
Could you do it again(`make uninstall`, `make clean`, `cmake ..`, `make` and `make install`) and post the contents of console here?

eokeeffe, 2019-03-13:
$ cmake ..
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Configuring Boost C++ Libraries...
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.64.0
-- Found the following Boost libraries:
--   thread
--   system
--   chrono
--   date_time
--   atomic
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.11") found components:  doxygen dot 
-- Configuring GNU Radio C++ Libraries...
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11git
 * INCLUDES=/opt/gnuradio/include
 * LIBS=/opt/gnuradio/lib/libgnuradio-runtime.so;/opt/gnuradio/lib/libgnuradio-pmt.so
-- Found GNURADIO_RUNTIME: /opt/gnuradio/lib/libgnuradio-runtime.so;/opt/gnuradio/lib/libgnuradio-pmt.so  
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: PMT
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11git
 * INCLUDES=/opt/gnuradio/include
 * LIBS=/opt/gnuradio/lib/libgnuradio-runtime.so;/opt/gnuradio/lib/libgnuradio-pmt.so
-- Found GNURADIO_PMT: /opt/gnuradio/lib/libgnuradio-runtime.so;/opt/gnuradio/lib/libgnuradio-pmt.so  
GNURADIO_PMT_FOUND = TRUE
-- Configuring LimeSuite C++ Libraries...
-- Checking for module 'LimeSuite'
--   Found LimeSuite, version 19.01.0-ga58636db
-- Found LimeSuite: /usr/local/include/lime, /usr/local/lib/libLimeSuite.so
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- Found SWIG: /usr/bin/swig3.0  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable version "2.7.12", minimum required is "2") 
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/evan
$ ccmake .

$ make -j4
Scanning dependencies of target pygen_python_dd9b7
Scanning dependencies of target _limesdr_swig_doc_tag
[... 221 lines cut ...]
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/api_8h.html
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/sink_8h__dep__incl.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/namespacegr.js
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/sink_8h__dep__incl.map
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/classgr_1_1limesdr_1_1sink.js
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/source_8h__dep__incl.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/folderclosed.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/source_8h__incl.md5
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/namespacegr.html
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/splitbar.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/source_8h__dep__incl.map
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/inherit_graph_1.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/modules.html
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/classgr_1_1limesdr_1_1sink__impl__inherit__graph.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/navtree.js
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/dir_97aefd0d527b934f1d99a682da8fe6a9_dep.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/source_8h.html
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/closed.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/classgr_1_1limesdr_1_1source.js
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/doxygen.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/api_8h__dep__incl.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/classgr_1_1limesdr_1_1source__impl.js
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/dynsections.js
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/sync_on.png
-- Installing: /opt/gnuradio/share/doc/gr-limesdr/html/doxygen.css

gracid, 2019-03-13:
Hmm, everything looks normal, can you try doing `sudo ldconfig` and checking if the issue persists?

eokeeffe, 2019-03-13:
Just tried that, tring the ffm_rds.grc file I get

self.limesdr_source_0 = limesdr.source('', 0, '')
TypeError: make() takes exactly 30 arguments (3 given)

gracid, 2019-03-13:
I'm not too familiar with nuances of installing in /opt/. Could you try checking if there are header files in /usr/local/include/limesdr/. You can try using this command:
`[ -f "/usr/local/include/limesdr/source.h" ] && echo "exists" `
If it does exist try doing `cmake --DCMAKE_INSTALL_PREFIX=/usr/local ../` and `sudo make install`

gracid, 2019-03-13:
Cool, closing the issue.

closed 2019-03-13 by gracid

eokeeffe, 2019-03-13:
Thanks again gracid :)