myriadrf/gr-limesdr #40: Debian Package?

andyl, 2019-04-04:
Wondering if there are any debian packages to install gr-limesdr.  I can compile from source, but prefer to use to package for automated installation.

gracid, 2019-04-05:
Currently there arent't any debian packages for gr-limesdr. The only alternative would be using snap package limesdr-grc [https://myriadrf.org/news/snap-packages-limesdr/](url)

9600, 2019-04-05:
@gracid can I request that we create Debian packaging for this. We do as of very recently have a snap with GNU Radio + gr-limesdr, but this won't let people add custom out-of-tree GR blocks and this is a very common requirement. 

We don't have a mechanism to distribute package builds for Debian at present, but we should have gr-limesdr packages distributed for Ubuntu via our GNU Radio PPA:

https://launchpad.net/~myriadrf/+archive/ubuntu/gnuradio

Note that the actual GNU Radio packages in there haven't been updated in some time and I'm not sure whether they will be or whether everyone just uses the main distro packages now. @csete could likely advise. Note that we also have LimeSDR packages in:

https://launchpad.net/~myriadrf/+archive/ubuntu/drivers

I did wonder whether we might put gr-limesdr in there as it is a driver of sorts, but then it provides GR blocks, so...

gracid, 2019-04-05:
Looking at it currently ubuntu 18.04LTS ships with 3.7.11-10 [link](https://packages.ubuntu.com/search?keywords=gnuradio), which is the same version in the myriadrf repository (though only for ubuntu 17.10 and I don't know whether they have to match exactly).

Personally, I think the package should be put in [https://launchpad.net/~myriadrf/+archive/ubuntu/gnuradio](gnuradio) repository as that's where modules such as gr-fcdproplus, gr-iio are located.

As for Debian packages, I think you can request them as there weren't any major issues since v2.0.0

gracid, 2019-06-17:
Debian packaging added with 47c6d40d369efc68f3f0d19d5a0105cd2d68732b commit.

closed 2019-06-17 by gracid