myriadrf/gr-limesdr #41: No samples returned when sampling rate <= 2.5M

heliochronix, 2019-04-09:
Hello,

I seem to be experiencing an issue (on two different hosts/archirectures, using two different LimeSDR USBs) where if the sample rate is set to 2.5e6 or lower then no samples are returned. This was experienced with any use of the LimeSDR RX block with sample rate <=2.5e6. Even some of the example flowgraphs (for example, FM_receiver.grc) do not work because the example sample rates are only 2M by default. However, if the sample rate is set to anything greater than 2.5M (such as 2.51e6) then suddenly samples begin to flow and everything works fine.

Software versions in use:
`LimeSuiteGUI 19.01.0 (commit 7557e29)`
`gr-limesdr 2.0.1 (commit 4125e1f)`

Output from flowgraph when it does not work:

```
---------------------------------------------------------------
LimeSuite Source (RX) info

##################
Connecting to device
##################
Devices found: 2
##################
Device list:
Nr.:|0|device:|LimeSDR-USB, media=USB 3.0, module=FX3, addr=1d50:6108, serial=0009070602500E37
Nr.:|1|device:|Remote, media=TCP, module=Z_Remote
##################
INFO: device_handler::open_device(): no serial number. Using first device in the list. Use "LimeUtil --find" in terminal to find prefered device serial.
##################
Device number 0 from the list is used.
RemoteControl Listening on port: 5000

Reference clock 30.72 MHz
##################

INFO: device_handler::enable_channels(): SISO CH0 set for device number 0.
INFO: device_handler::set_samp_rate(): set sampling rate: 2.5 MS/s.
INFO: device_handler::set_rf_freq(): RF frequency set [RX]: 436 MHz.
RX LPF configured
INFO: device_handler::set_analog_filter(): INFO: device_handler::set_gain(): set gain [RX] CH0: 50 dB.
INFO: device_handler::set_antenna(): CH0 antenna set [RX]: LNAL.
Rx calibration finished
INFO: device_handler::calibrate(): INFO: source_impl::init_stream(): source channel 0 (device nr. 0) stream setup done.
```

If there's any additional info that could help diagnose this problem please let me know and I will do my best to supply it. Thank you for your time and efforts!

gracid, 2019-04-12:
This is LimeSuite library issue (LMS_RecvStream returns 0 samples when sample rate is < 2.5M, tested with basicRX example)

gracid, 2019-04-12:
Should be fixed at https://github.com/myriadrf/LimeSuite/commit/357ad5dd0d71304179d476b38e67240527d917df just update your LimeSuite.

closed 2019-04-12 by gracid

heliochronix, 2019-04-12:
Appreciated, thank you.