myriadrf/gr-limesdr #42: Add VCTCXO DAC parameter in sink and/or source blocks

JDat, 2019-04-27:
It would be great if there is parameter on sink and/or source block to manually (or semiautomatically) control VCTCXO DAC parameter from GNU radio by user.

PS: I am no programmer, but will try to add this as personal exercise and play with code.

gracid, 2019-05-30:
The problem I'm having about this is that this VCTCXO DAC parameter is written to flash memory and saved even after power off. Misusing it or forgetting about it could cause some severe problems in the future and finding out about it would be difficult.

The idea I've had is to have the ability to change this parameter similarly to how [change_fifo_size](https://github.com/myriadrf/gr-limesdr/blob/master/lib/source_impl.h#L91) is handled - not in the grc blocks themselves but only through (generated or written) python code modification.

ztamosevicius, 2019-05-30:
As an addition to @gracid comment, VCTCXO DAC value change is a service procedure, not application like GNU Radio. VCTCXO DAC value change needs to be done via LimeSuiteGUI - ideally.

JDat, 2019-05-30:
What if I need to compensate this value on the fly while working on Es'Hail-2?

gracid, 2019-05-30:
Like I've said, it could be implemented similarly to change_fifo_size, thus making sure it isn't very easily accessible and only people who know what they are doing would be able to use it.

gracid, 2019-06-04:
VCTCXO DAC value modification added in 18970de4daac419655752cce1deab80adedf46af. To allow changes to this parameter "Allow TCXO DAC control" in the "Advanced" tab must be set to "Yes".

closed 2019-06-04 by gracid