myriadrf/gr-limesdr #44: Upgrade gr-limesdr to support GNU-Radio 3.8

gracid, 2019-06-04:
There will be a lot of changes necessary to support GNU-Radio 3.8, currently it is still in [development](https://github.com/gnuradio/gnuradio/projects/1). Documentation to port to 3.8 is provided  [here](https://wiki.gnuradio.org/index.php./GNU_Radio_3.8_OOT_Module_Porting_Guide).

Right now it doesn't compile(#37) and the block description needs to be changed to yaml(#27).
At minimum, things that will need to be updated/changed are: CMake, python?, xml to yaml, examples.

mormj, 2019-08-19:
Now that 3.8 is out, is this work currently in progress?

mormj, 2019-08-19:
If not, I'd be happy to help out

gracid, 2019-08-23:
Right now it isn't being worked on, it might be picked up on september.

gracid, 2019-08-29:
It seems that basic RX is working (tested on clean ubuntu 18.04 with gnuradio maint-3.8 branch)

eblot, 2019-08-29:
> It seems that basic RX is working (tested on clean ubuntu 18.04 with gnuradio maint-3.8 branch)

did you try with Python 3.x?

gracid, 2019-08-29:
I only checked with python3 and only tested whether basic signal receive was working.

eblot, 2019-08-29:
Great!

9600, 2019-09-30:
@gracid any update on when we'll have a release with full support?

gracid, 2019-10-09:
I would say release would be coming next month since there were some issues.

gracid, 2019-11-15:
The release will happen simultaneously with LimeSuite realease, the branches will probably stay as they are.

ra1nb0w, 2019-11-15:
ok. thank you

gracid, 2020-01-28:
Made the release [v3.0.1](https://github.com/myriadrf/gr-limesdr/releases/tag/v3.0.1)

closed 2020-01-28 by gracid