myriadrf/gr-limesdr #46: Receiving on both channels simultaneously

fwfmc, 2019-06-18:
Hi all,

I am currently getting into Software Defined Radio using the LimeSDR USB Board with GNU Radio. For a research project, I want to be able to transmit a signal over one antenna at around 2.4GHz and receive it over two different antennas (best case: on the same board / okay case: on another board).
I've created a simple flow graph, receiving a transmitted signal over either channel A or B and feeding it into a Time Sink block - which, from my perspective, works as expected. But as soon, as I want to use a LimeSuite Source working on both channels (A+B)MIMO, no signal is received at both channels. 

- Do I additionally have to tell the board or the flow graph that it is used in MIMO mode now, except changing the LimeSuite Source's channel to (A+B)MIMO?

- I know that my usecase is rather SIMO than MIMO, but this shouldn't affect the receiver side, should it?

- Is receiving on two antennas on channels A and B even possible with the LimeSuite Source or do I basically misunderstand the concept of "channels"?

Thank you in advance!

gracid, 2019-06-19:
Hello, 

To answer your questions.
1. No, setting channel to (A+B) MIMO is enough.
2. You can't use SISO on TX side and MIMO on RX side on the same device, though feeding null data to one channel might work. 
3. Yes, it is possible. [Here](https://pastebin.com/zN8r1R27) is a simple MIMO loopback example. 

For further discussion on how to use the gr-limesdr blocks or LimeSDR board please go to the MyriadRF [forum](https://discourse.myriadrf.org/). 

Closing as this is a non issue with gr-limesdr.

closed 2019-06-19 by gracid

fwfmc, 2019-06-19:
Thank you for your fast response!
Unfortunately the pastebin link is not valid, could you fix that?

gracid, 2019-06-19:
Pastebin link is working for me on multiple devices. Save the text as .grc file and open it with GNU Radio.

fwfmc, 2019-06-19:
I'm sorry, the problem was caused by my proxy settings. Thank's for your help!