myriadrf/gr-limesdr #50: Build error with gnuradio home made (3.9.0.0-git)

Artemia76, 2019-09-14:
I get 
`gr-limesdr/swig/limesdr_swig.i:5: Error: Unable to find 'gnuradio.i'
make[2]: *** [swig/CMakeFiles/limesdr_swig_swig_compilation.dir/build.make:64: swig/CMakeFiles/limesdr_swig.dir/limesdr_swigPYTHON.stamp] Error 1
make[2]: *** Suppression du fichier « swig/CMakeFiles/limesdr_swig.dir/limesdr_swigPYTHON.stamp »
make[1]: *** [CMakeFiles/Makefile2:309: swig/CMakeFiles/limesdr_swig_swig_compilation.dir/all] Error 2
`
But in /usr/local/include/gnuradio/swig/ the gnuradio.i is present

sniporbob, 2019-09-17:
I second this. Having the same issue. I'm using Arch linux and installed gnuradio 3.8.0.0-2 via pacman. My /usr/local/ folder does not contain anything related to gnuradio but /usr/include/gnuradio/swig contains gnuradio.i

gracid, 2019-09-17:
Currently, the master branch will be used with gnuradio3.7 until ubuntu ships with 3.8. To use gr-limesdr with gnuradio3.8 please change branch to gr-3.8.

closed 2019-10-09 by gracid

cscheidl, 2019-11-21:
I've a debian08 system and first tried the gr-3.8 branch gnuradio3.9  -- with the above described failure. Then I switched back to the distro's gnuradio3.7 by using apt install --reinstall gnuradio. Still the same failure. I propose to reopen this issue. It isn't solved....