myriadrf/gr-limesdr #53: run FM_receiver.grc,  yaml.scanner.ScannerError

jin-github102, 2019-11-01:
How to deal with it??

dell@dell-XPS-15-9550:~$ gnuradio-companion
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.9.0.0-git >>>

Block paths:
	/usr/local/share/gnuradio/grc/blocks
Warning: restarting the docstring loader (crashed while loading 'qtgui_bercurve_sink')
Warning: restarting the docstring loader (crashed while loading 'qtgui_const_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_edit_box_msg')
Warning: restarting the docstring loader (crashed while loading 'qtgui_freq_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_histogram_sink_x')
Warning: docstring loader crashed too often

Loading: "/home/dell/Downloads/FM_receiver.grc"
Error: mapping values are not allowed here
  in "/home/dell/Downloads/FM_receiver.grc", line 117, column 71
>>> Failure
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/MainWindow.py", line 264, in new_page
    file_path=file_path,
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/gui/Notebook.py", line 106, in __init__
    initial_state = flow_graph.parent_platform.parse_flow_graph(file_path)
  File "/usr/local/lib/python3.6/dist-packages/gnuradio/grc/core/platform.py", line 337, in parse_flow_graph
    data = yaml.safe_load(fp)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 94, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 72, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 35, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 58, in compose_document
    self.get_event()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 118, in get_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 193, in parse_document_end
    token = self.peek_token()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 128, in peek_token
    self.fetch_more_tokens()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 220, in fetch_more_tokens
    return self.fetch_value()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 576, in fetch_value
    self.get_mark())
yaml.scanner.ScannerError: mapping values are not allowed here
  in "/home/dell/Downloads/FM_receiver.grc", line 117, column 71

gracid, 2019-11-15:
**Welcome to GNU Radio Companion 3.9.0.0-git** Can you try using main-3.8 branch?

jin-github102, 2019-11-16:
Yes i can. Is this error because of version?

gracid, 2019-11-18:
GNURadio 3.9 is currently not supported.

gracid, 2020-01-08:
Closing stale issue.

closed 2020-01-08 by gracid