myriadrf/gr-limesdr #55: ran examples, fm_receiver, no sound

jin-github1002, 2020-01-22:
hello , I ran examples, fm_receiver. I can see a impulse in the graph. But there seems to be some noisy instead of the fm sound. Does anybody help to find the problem? Thanks
[image]
[image]
[image]

jin-github1002, 2020-01-22:
I am in Texas, near Dallas.

gracid, 2020-01-24:
You are getting no sound because of alsa driver is reporting audio underruns("aU" in the message log) meaning your CPU wasn't able to process the samples in time. You could try to lower the sampling rate keeping in mind that other parameters will need to be changed.

jin-github1002, 2020-01-24:
Do you have any suggestion about the sample rate? I have no idea what it
should be.

I check cpu info as follows:

# cat /proc/cpuinfo | grep name | cut -f2 -d: | uniq -c
  8  Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz

jin-github1002, 2020-01-26:
hello, I have tested much. But i failed without knowing the reasons.

1)somebody says work is still in progress to port LimeSuite to GR3.8. Is it OK? 
Do i need to use GR3.7? Is there any other software versions that i need to change?
i use GNURadio 3.8.0.0
gr-limesdr 3.0.0
LimeSuite: 19.04

2)another question, i have changed samp_rate for many times. for example, 192k, 1m, etc. But it is the same as before. Would you please give a possible samp_rate that is ok?

3) when i use samp_rate=2M, in rational resampler: Decimation=200, interpolation=48, 
I think 2*48/200=480k, it is just the input as quatrature rate in WBFM. Does the quatrature rate need to be equal to the output of rational resampler? 

Thank you so much!

gracid, 2020-01-28:
With that CPU it there shouldn't be a problem processing the samples. To answer your questions:
1. Yes, it is okay, the only thing gr-3.8 branch is missing is a release tag.
2. The default value of 2M should be working fine on most systems.
3. Yes, the quadrature rate needs to be equal to the output of resampler and WBFM output should match audio sink.

Things to try out:
1. I see that I've left the default value of "ok to block" in the audio sink, set it to "No".
2. Try different audio sink rates, for that you would need to change rational resampler and quadrature values (44.1k example: Decimation=2000, Interpolation 441, Quadrature rate: 441e3).

ChristopheSeguinot, 2020-01-28:
Hi 

I've been working at using LimeSDR Mini under NGNURadio 3.8 After many attempt, I succeeded using: 

- GNURadio 3.8 (build from master branch, but any 3.8 .deb package should work too)

- gr-limesdr , gr-3.8 branch from https://github.com/myriadrf/gr-limesdr.git

I attached the zipped .grc file
About your flowgraph: 
- 48 kHz can be too much for several sound card, try using 44.1k Hz or lower (see audio source block documentation). 
- the L+R audio channel generally exhibit some DC offset. It's better to remove it with a DC blocker. 

Regards, Christophe

[image]
[limesdr_wbfm_receive.grc.zip](https://github.com/myriadrf/gr-limesdr/files/4122474/limesdr_wbfm_receive.grc.zip)

jin-github1002, 2020-01-28:
thanks @gracid @ChristopheSeguinot 
i will try now.
@ChristopheSeguinot i saw you said : you used LimeSuiteGui to calibrate the LimeSDR Mini receiver. Don't know if that step was requested, but it now works!  
would you please tell me what you did (or the steps by steps) to calibrate the LimeSDR?

jin-github1002, 2020-01-29:
@gracid 
i install gr-3.8 now. but i have some errors. Could you give some recommend to solve it?

dell@dell-XPS-15-9550:~/gr-limesdr/build$ cmake ../
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- Found VOLK: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- LimeSuite version 19.04.1-ga5b3a10f found.
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen dot 
-- Using install prefix: /usr/local
-- Building for version: 3.0.1.0 / 3.0.1
-- No C++ unit tests... skipping
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
-- Found SWIG: /usr/bin/swig3.0  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found version "3.6.9") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dell/gr-limesdr/build
dell@dell-XPS-15-9550:~/gr-limesdr/build$ 
dell@dell-XPS-15-9550:~/gr-limesdr/build$ 
dell@dell-XPS-15-9550:~/gr-limesdr/build$ 
dell@dell-XPS-15-9550:~/gr-limesdr/build$ make 
Scanning dependencies of target gnuradio-limesdr
[  6%] Building CXX object lib/CMakeFiles/gnuradio-limesdr.dir/device_handler.cc.o
[ 12%] Building CXX object lib/CMakeFiles/gnuradio-limesdr.dir/source_impl.cc.o
[ 18%] Building CXX object lib/CMakeFiles/gnuradio-limesdr.dir/sink_impl.cc.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/liborc-0.4.so', needed by 'lib/libgnuradio-limesdr.so.3.0.1.0'.  Stop.
CMakeFiles/Makefile2:135: recipe for target 'lib/CMakeFiles/gnuradio-limesdr.dir/all' failed
make[1]: *** [lib/CMakeFiles/gnuradio-limesdr.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

gracid, 2020-01-29:
I'm curious as to how you were able to use gr-3.8 before this installation but running `sudo apt-get install liborc-0.4-0 liborc-0.4-dev` should fix it.

jin-github1002, 2020-01-29:
I will check how i installed gr-3.8 before.
this time i reinstalled gr-3.8 as you recommended yesterday. it is ok. 
but when i running example from @ChristopheSeguinot  ,there is an error. could you give some recommend? @gracid 

Loading: "/home/dell/Desktop/limesdr_wbfm_receive.grc"
>>> Done

Generating: '/home/dell/Desktop/uhd_wbfm_receive.py'

Executing: /usr/bin/python3 -u /home/dell/Desktop/uhd_wbfm_receive.py

Traceback (most recent call last):
  File "/home/dell/Desktop/uhd_wbfm_receive.py", line 38, in <module>
    import limesdr
ModuleNotFoundError: No module named 'limesdr'

>>> Done (return code 1)

my version:
ubuntu: 18.04
gnuradio:3.8.0.0
gr-limesdr: gr-3.8 dont know the version
LimeSuite: v19.04.1-ga5b3a10f

and i used LimeSDR USB not LimeSDR mini (different from @ChristopheSeguinot )

jin-github1002, 2020-01-29:
i can run gnuradio and .grc now. 
but the .grc file does not work well on my laptop. @ChristopheSeguinot i just changed the freq to a fm here. Could you give some comments?

gracid, 2020-03-27:
Closing a stale issue.

closed 2020-03-27 by gracid