myriadrf/gr-limesdr #56: build failure

sjoerdtimmer, 2020-02-06:
The new gr-3.8 fails to compile for me.

Compilation output:
https://pastebin.com/9c94ivdg

System:
archlinux x86_64
python 3.8.1
python2 2.7.17
swig 4.0.1
limesuite 20.01.0
gnuradio 3.8.0 (actually 3.8.0.0 due to archlinux packaging mistake)

Can you recommend a specific commit version of gnuradio that this was tried and tested with?

gracid, 2020-02-07:
The last GNURadio commit I was testing with is https://github.com/gnuradio/gnuradio/commit/2f5c81d5552b68700da09a508ced8995a3ecb512

Though the problem isn't really from gnuradio version here. It looks like you had built a pretty old version of gr-limesdr before and swig isn't happy about that. Can you try removing build directory and try compiling completely from scratch?

sjoerdtimmer, 2020-02-07:
I'm building the gr-3.8 branch. commit bf6bcf4 to be exact.

I see that I only pasted the error. Here is a full clean build log including the git checkout:
https://pastebin.com/CKz8Nxsn

gracid, 2020-02-07:
I've tested this on ubuntu 18.04 with the same swig and gnuradio versions and manjaro 18.1.5 which has the same package versions as you are using, both seem to be working fine.

This is one weird error though, swig somehow manages to get year-old function declarations and I can't really tell how and why. There is AUR package for [gr-limesd](https://aur.archlinux.org/packages/gr-limesdr-git/)r, can you try to see if that builds successfully?

sjoerdtimmer, 2020-02-07:
I found the issue:
The old gr-limesdr aur package had installed the swig interface globally and that was being used instead of the one generated from the sources in the package. 

In case anyone else runs into this:
pacman -Rscn gr-limesdr before compiling solves the issue

closed 2020-02-07 by sjoerdtimmer