myriadrf/gr-limesdr #61: What is the max gain?

hopkinskong, 2020-03-18:
It seems that the limesdr block limited max gain to 60dB:
[image]

However, when I was using SDRangel to output some NFM signals, it seems it can go to 70dB.
[image]

I can also confirm that setting 70dB in SDRangel had higher power output than 60dB in GNURadio.

What is the actual max gain for Tx?

gracid, 2020-03-26:
I have update the maximum gain values to match LimeSuite with bf7e7d375ec95a5d24d14dea9008b5151b47249c commit. Right now they are in range of [0,73] dB, it is possible to bring them even higher but I won't get into that.

kconkas, 2020-06-21:
@gracid the max sink gain is still limited to 60dB in https://github.com/myriadrf/gr-limesdr/blob/master/grc/limesdr_sink.xml#L442 and https://github.com/myriadrf/gr-limesdr/blob/master/grc/limesdr_sink.xml#L445. Is this intentional?

9600, 2020-08-03:
@gracid could you clarify, thanks.

gracid, 2020-08-04:
Should be fixed with d7ba72128060158fc819fa4942fd3797bf501f31.

closed 2020-08-04 by gracid