myriadrf/gr-limesdr #64: Clarify behaviour when using an INI file

9600, 2020-07-28:
See:

https://discourse.myriadrf.org/t/limesdr-usb-inconsistent-configuration-and-sample-rate/6345/3

Perhaps we could grey out all the other settings when an INI file is loaded?

gracid, 2020-07-28:
I think the solution to this problem would be to just update documentation. Switching different block elements can get quite messy and annoying to work with.

9600, 2020-07-28:
Sure, sounds good and as long as the behaviour is clear.

9600, 2020-08-03:
I can see the notes in the Documentation tab, but I'm not sure that the behaviour is still sufficiently clear.

One option might be to move the File field to lower down on the General tab, so that it's set apart from the Serial and Channel etc. fields and not located in the middle of them. Then directly above or below the File field, place a warning to note that loading a file will override other settings.

gracid, 2020-08-04:
Sadly, you cannot just place a warning note in middle of parameters or pretty much anywhere else in the block besides the documentation tab. I also think the location of File field is good because in order it goes: device identification(serial), File field (most important setting, overriding other ones), channel configuration (whether the block will need to use MIMO and which channel to use).

gracid, 2020-08-26:
INI file load has been moved to advanced tab in 0b9a6d8c106f5bad4b7adc248b67382051e089a2.

closed 2020-08-26 by gracid