myriadrf/gr-limesdr #75: Support for GNURadio 3.9.x

eggelene, 2021-05-22:
I tried to use the latest LimeSDR ( from ppa:myriadrf/gnuradio) sink in GNURadio 3.9.1.0. But it seems that the sink is not compatible with latest GNURadio release. 

Is there any plan to add LimeSDR support for GNURadio 3.9.1.0 ? Preferable installable through the ppa:myriadrf/gnuradio.

Phil67ago, 2021-06-05:
I have the same question.
gr-3.9 has replaced SWIG with Pybind11.
I made some efforts following this guid but I'm kind of lost when it comes to adding binding at step 4.
Call gr_modtool bind for each block in your OOT.
https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide

It would be nice if someone created a branch gr-3.9 and made the basic porting steps and CMake stuff
as a starting point.

chrisjohgorman, 2021-06-17:
I have tried to migrate this module to gnuradio 3.9.  My attempt can be found at https://github.com/chrisjohgorman/gr-limesdr.git .
I don't currently own a limesdr to test my modifications out on, but it builds on my system and installs.  I can import the module from python.  If you find bugs, and I'm sure there are some (rfe boards may not be properly supported.)  Please email me and I will try to hunt them down.

Phil67ago, 2021-06-26:
Ok Fine! I will give it a try. It may take me a couple of weeks before I can dig into it...

chrisjohgorman, 2021-06-26:
@Phil67ago please re download gr-limesdr.git from my repository.  I forgot to do a git add to add the new files and as a result my github code was incomplete.  I have tested it and it _should_ work now.  Sorry for the omission.

NZSmartie, 2021-06-28:
Hey @chrisjohgorman I gave your fork a go, I fixed a few things and created the needed python bindings.
I can confirm that my LimeSDR-mini now works with the source module in GNU Radio 3.9, however I have not tested the sink, yet. As I don't have a project ready to go.

dsgibbons, 2021-09-15:
I tested @chrisjohgorman fork - seems to work well for LimeSDR-USB. Thanks for doing this.

wkazubski, 2022-03-06:
Please bump the library version declared in CmakeFile.txt. Last release for GR-3.7 was anounced as 2.0, last snapshot in 3.8 branch has  library version 3.1.5, so version for GR-3.9 should be 3.2.0 or even 4.0.0 to avoid confusion during upgrade.

chrisjohgorman, 2022-03-06:
Hi Wojciech,

If I understand you, I need to update the VERSION_MAJOR, VERSION_API and
VERSION_ABI to something greater than 3.1.5.  I believe it should be 3.2.0,
but I don't know what lime microsystems would prefer.  I'm on my way to
change it and will set it to 3.2.0.  I hope this helps.

Take care.

Chris

wkazubski, 2022-03-06:
Yes. . If there is no major changes, version 3.2.0 is OK. I suppose that version 1.0.0 is a default setting in Cmake. Leaving it nay make some confusion for package managers in various Linux distributions. For example openSUSE / SUSE Linux package libries separatedly and require library package to have the same version as the library executable  inside.

Wojciech

chrisjohgorman, 2022-03-06:
Yes. 1.0.0 is the default setting of gr_modtool which is used to make the
module.  Good to hear that 3.2.0 is OK.  Take care.

Chris