myriadrf/gr-limesdr #80: Lime SDRmini and gnuradio

wecourduff, 2022-04-20:
[image]

[image]

This is the transmitter and receiver for two limesdr mini's.  I have gotten rid of the static on the Audio sink, however I am not hearing the voice from the .wav file on the transmitter at the receiver.  Any suggestions on how I can hear the voice message at the receiver?

9600, 2022-04-20:
Have you posted details to the forum?

https://discourse.myriadrf.org/c/projects/limesdr/21

wecourduff, 2022-04-20:
> Have you posted details to the forum?
> 
> https://discourse.myriadrf.org/c/projects/limesdr/21

Just posted it, thank you

9600, 2022-04-20:
Closing this issue for now as it's far more likely to be a flowgraph configuration error than an issue with gr-limesdr. We can reopen if this turns out not to be the case.