myriadrf/gr-limesdr #86: LimeSDRmini 2.0 and GNUradio

MichTri, 2023-02-13:
Hi,
I don´t get the limeSDR working together with GNUradio, either with version 3.10 nor with 3.7.
In 3.10 the limesdr_sorurce block is not recognized ("Missing Block").
In 3.7 I am able to execute the flow graph, but the output baseband and RF spectrum shows only a flat line. Also the green LED in the corner near the RX-Antenna is off.

What I am missing?
Thanks for any help.

9600, 2023-02-13:
How did you install Lime Suite? For the time being you need to install from source and build from git master, until a new release is tagged with LimeSDR Mini v2 support.

MichTri, 2023-02-13:
I have used the LimeSuiteGUI.exe within the PothosSDR/bin folder.
But my problem is not the LimeSuite. My Problem ist to get the limeSDRmini V2.0 working together with GNUradio.

9600, 2023-02-13:
> But my problem is not the LimeSuite. My Problem is to get the limeSDRmini V2.0 working together with GNUradio.

gr-limesdr depends upon Lime Suite and hence you must have the correct version of Lime Suite library installed, otherwise it won't work. If you've installed LimeSuiteGUI from the PothosSDR installer, this also won't work, as the installer has not been updated yet and so will use an old version of Lime Suite.

You will need to:

1. Build Lime Suite (git master) from source and install
2. Build applications and link against the above

closed 2023-02-13 by 9600