myriadrf/gr-limesdr #87: LimeSDR-Mini v2.2 does not start immediately

nedsana, 2023-04-03:
Hi, 
When I run the attached grc schematics, initially no IQ data is being processed. It seems like RX is not running. I found that I need to toggle the analog filter GUI field (simply add or subtract 1 from the initially set value) and everything start working fine. 

I think the issue is related to LimeSDR-Mini v.2.2, because previously I had used LimeSDR-Mini v.1.<something> and LimeSDR-USB v.1.4 and there I have not faced such problems.

My setup:
LimeSDR-Mini v.2.2
Ubuntu 20.04
GnuRadio 3.8
gr-limesdr: branch gr-3.8, installed from source

The same problem can be seen on:
LimeSDR-Mini v.2.2
Ubuntu 22.04
GnurRadio 3.10
gr-limesdr 3.0.1 installed from apt

[SyncedLimeSDR.zip](https://github.com/myriadrf/gr-limesdr/files/11136359/SyncedLimeSDR.zip)

ztamosevicius, 2023-04-03:
Are you using Rx data stream only? If yes, then you have to enable Tx channel as well for v2.x versions.

nedsana, 2023-04-03:
In this very grc, I transmit from LimeSDR-USB, and indeed, LimeSDR-Mini v.2.2 is used for RX only. 
But how should I enable the Tx channel? In gnuradio I don't see such option. Should I use LimeSuiteGUI or something else?

nedsana, 2023-04-03:
I kept searching and found the following articles, which explained the issue a bit better:
https://github.com/AlexandreRouma/SDRPlusPlus/issues/1025
https://discourse.myriadrf.org/t/limesdr-mini-v2-reading-samples-from-rx-fifo/7900

So in GnuRadio, for block "LimeSDR Source (RX)", I changed the property "Channel" from "A" to "A+B". Running the grc resulted in error "INFO: device_handler::enable_channels(): Invalid channel number", which seems to come from LimeSuite:
inline lime::LMS7_Device* CheckDevice(lms_device_t* device, unsigned chan). There the call to 
lms_dev->GetNumChannels() is hardcoded to return 1, which always causes error, if A+B channel is set in GnuRadio.

Am I missing something?

My version of LimeSuite is v22.09.0 - stable. May be I should use another?

nedsana, 2023-04-04:
Hi, please ignore my previous comment. I just scarcely looked at the code, when I wrote it. 

I solved my problem, based on [ztamosevicius](https://github.com/ztamosevicius) comment.

I had to update the implementation, so I added the following method:

`bool device_handler::is_limesdr_v2(int device_number)
{
  const lms_dev_info_t* info =  LMS_GetDeviceInfo(device_vector[device_number].address);
  return 0 == std::strcmp("LimeSDR-Mini_v2", info->deviceName);
}
`

to check if Limesdr-mini version 2 is used. Then in source_impl::source_impl I would create the TX channel if needed:

`        if (device_handler::getInstance().is_limesdr_v2(stored.device_number))
        {   //for LimeSDR v2 we need to enable the TX
            std::cout << "!!! DETECTED LIMESDR MINI VERSION 2 !!!" << std::endl;
            device_handler::getInstance().enable_channels(stored.device_number, stored.channel_mode, LMS_CH_TX);
        }
`

Thank you, closing issue.

closed 2023-04-04 by nedsana